Protocentral/protocentral_max30003 #3: Runing on arduino

ALEXRG27, 2017-05-26:
I'm trying to read the register with the value of the R_R , but I can't read anything, I get this :
ú            0,0,0,,0

     Serial.print(RTOR_msb);
     Serial.print(",");
     Serial.print(RTOR_lsb);
     Serial.print(",");
     Serial.print(rtor); 
     Serial.print(",");
     //Serial.print(rr);
     Serial.print(",");
     Serial.println(HR);   
How can I Read the ECG value from the arduino?
I also have a problem with the code, when i compile it says:
sketch\MAX30003.h:28:0: warning: "NO_OP" redefined

 #define   NO_OP           0x7F

 ^

sketch\MAX30003.h:9:0: note: this is the location of the previous definition

 #define   NO_OP           0x00

VenkateshBhat, 2017-05-27:
Dear Alex,

- Are you able to see the waveform on the gui?, could you attach the screenshot?
- If the data is clean these registers rr and HR will hold the corresponding values.
- NO_OP register doesn't have any impact on the operation, the warning can be ignored.
- Please ensure that you have the latest version of Arduino IDE

ALEXRG27, 2017-05-29:
No I can't see anything->
[image]

[image]

ALEXRG27, 2017-05-29:
If i disconect the board, i get this :
[image]
So the board i getting something but I don't know why i can't read the pulse

VenkateshBhat, 2017-05-29:
Thanks for sharing, are you sure about the wiring connections?,  do check with the latest version of Arduino IDE -- http://www.arduino.org/downloads

VenkateshBhat, 2017-05-29:
Please attach the picture of Hardware setup too.

ALEXRG27, 2017-06-07:
I've checked my board and all seems to be good. I used the same layout as you.

[image]

VenkateshBhat, 2017-06-21:
Have you made any progress on this?, it seems to be fuss in the assembly.

ALEXRG27, 2017-06-23:
No, I've tested the whole board and everything seems fine, i'll test it on my final design without the arduino I wish I could  read the data from the sensor.

DJMBurkhard, 2017-09-27:
Hello,
I have two questions: 

1.) Connecting the max30003 board with Arduino Uno:
Should I use the 3V or the 5V?
I assume it is the one or the other? 
And the power is supplied via the computer? (I am using a MAC)

2.) I cannot upload the sketch to Arduino
I receive the message:
fatal error: TimerOne.h: No such file or directory
#include <TimerOne.h>

Where can I find this file? It seems to be missing? 
It is not in the folder, 
I see the MAX30003.h  only -

closed 2017-09-28 by VenkateshBhat

DJMBurkhard, 2017-09-29:
Thanks for your previous respond which helped and could get:
The graph I receive now, using processing is the attached image
[image]

- the BPM are jumping around between 3 and 60, around 60, that would be correct, but it is jumping everywhere and:
- the graph is much different to what you show in the manual, see above image
-  the cvs file contains just one symbol - no data.
- I am told that the " protocentral_max30003_gui.app" is defect, and should take it to the trash.