Protocentral/protocentral_max30003 #4: Analog voltage differences

fzldmn, 2017-06-10:
Hello in datasheet  recommended analog voltage is max 2V. 
But in your design u used 3.3V 

Will it be a problem for getting 
 stable measurements.

VenkateshBhat, 2017-06-21:
Yes, the recommended power supply voltage is 1.1V - 2V , we tested for the long duration and there is no heating as such. We are getting the stable readings.

DJMBurkhard, 2017-09-27:
Hello, 

1. Connecting the max30003 board with Arduino Uno: 
    Should I use the 3V or the 5V? 
    I assume it is one or the other? And the power is supplied via the computer (I am using a MAC)
   
2.   I cannot  upload the sketch to Arduino 
    I receive the message:
   fatal error: TimerOne.h: No such file or directory
   #include <TimerOne.h>

    Where can I find this file? It seems to be missing.

VenkateshBhat, 2017-09-28:
@DJMBurkhard 
1. Connect to 5V if you are using Arduino Uno.
2. Install the library externally for TimerOne.h
Go to sketch->Include libray -> Manage libraries 
[image]

closed 2017-10-09 by VenkateshBhat