Protocentral/protocentral_max30003 #6: BPM and Graphic with Processing strange

DJMBurkhard, 2017-10-05:
The graph I receive now, using processing is the attached image:
[image]

- the BPM are jumping around between 3 and 60, around 60, that would be correct, but it is jumping everywhere and:
- the graph is much different to what you show in the manual, see above image
-  the cvs file contains just one symbol - no data.
- I am told that the " protocentral_max30003_gui.app" is defect, and should take it to the trash.

VenkateshBhat, 2017-10-05:
@DJMBurkhard 
Thanks for the feedback, the signal is too erratic.
ensure: 
 > there is a good contact of electrodes on body 
 > have a latest version of the Arduino(1.6.11 or above)
 >  Remove the power adapter from the PC/laptop and check

I will check on the no data on .csv file and get back to you.

DJMBurkhard, 2017-10-05:
Thank you - 

So far, these issues are under control,
- I have the latest Arduino installed, 1.8.4,
- the electrodes are new and were placed as described,
- the power adapter was not plugged in while the trying to take measurement 
( I am using a mac book air)
I am not aware of any other possible disturbing equipment 
nearby: 

In fact, before, I had taken measurements with another peace of equipment ,
"the pulse sensor", which is just wrapped around the finger.
That worked fine, as along as I was not moving.

But my hope is still in protocentral_max 30003, and I expect it to be more accurate.
However, as said before,  the BPM is mainly far too low, 
at 3 to 30, rarely  around 60, which would be normal. 

So, if you have any other idea, that would be much appreciated -