Protocentral/protocentral_max30003 #7: Design

amir1994a, 2017-10-06:
In page 44 of the datasheet of max30003, in the recommended circuit it has used 1k resistor while you have
used 49.9k resistor? is this because of a particular reason? what will happen if you use a 1k resistor instead of 49.9k?

VenkateshBhat, 2017-10-07:
@amir1994a ,
the diagram mentioned in the datasheet is a typical application circuit not the recommended values.  The RC filter here acts as an anti-aliasing filter .Reference diagram is here https://www.maximintegrated.com/en/design/tools/appnotes/6312/MAXREFDES100_SCH_RA.pdf

amir1994a, 2017-10-07:
thank you, what about the zero value resistor?is it going to be a problem if we don't use them?

VenkateshBhat, 2017-10-07:
No, it does not affect the performance.

closed 2017-10-09 by VenkateshBhat

Tomisdebom, 2021-09-13:
> @amir1994a ,
> the diagram mentioned in the datasheet is a typical application circuit not the recommended values. The RC filter here acts as an anti-aliasing filter .Reference diagram is here https://www.maximintegrated.com/en/design/tools/appnotes/6312/MAXREFDES100_SCH_RA.pdf

Wait, the reference design you just posted is different than the schematics that are in the Github repo. Are you sure your reference design corresponds to the Protocentral board?