Protocentral/protocentral_max30003 #8: A question about the external clock

amir1994a, 2017-11-06:
Hi, I've been trying to understand the board and the code, I'm confused a little bit.
How do you produce the external clock?it seems that there is no crystal or oscillator on the board.
what is this line for?:
    digitalWrite( CLK_PIN, digitalRead(CLK_PIN ) ^ 1 ); // toggle Digital6 attached to FCLK  of MAX30003

why do toggle the fclk pin?

VenkateshBhat, 2017-11-06:
@amir1994a 
Thanks for the comment,
Since Arduino doesnot have internal RTC we are deriving using the timer with 64KHz clock frequency and toggling a GPIO.

amir1994a, 2017-11-06:
Thank you for your answer, at first my code(yours actually) wouldn't compile because of the lack of time1 library, i downloaded it from this link(9th version) and it compiled without error, is that the right procedure?

one more question, actually, i did not use your board because i couldn't find it in my country, so i tried to use your schematic and make it myself. only i did not use 0-ohm resistors. and i did not use ground plane(except for the ic pad). and i used this voltage converter:

https://www.sparkfun.com/products/12009

my board did not work, i assumed that maybe this voltage converter is not good for spi connection and maybe 5 and 1.8 have a lot difference for my voltage converter.
I'm really frustrated and i know that only you can help me.
will you please help me a little bit?
(also the ac line frequency is 50hz in my country and is different from 60hz of some countries i don't know that this will cause any problem or not)

Thanks a million
please help me!!!

VenkateshBhat, 2017-11-07:
@amir1994a ,
Do not worry we are here to help you out!
Could you please share your design? what signal are you getting at the output? FCLK is critical are you sure about the signal? the level converter could not be the problem, BSS138 is good enough.

amir1994a, 2017-11-07:
Hi, thank you, you are so kind.
Actually, I get no signal at the output.
I have attached a pdf file of my schematic and PCB, I'll be grateful f you check it out. it is a little bit messy!
Can you explain a little bit about TimerOne library?, because arduino does not seem to have this library by default.
I'll be thankful if you tell me what my problem is.
[heart.PDF](https://github.com/Protocentral/protocentral_max30003/files/1448888/heart.PDF)

amir1994a, 2017-11-12:
Hi there,
I was wondering whether you had the time to look at my previous question.
I'm enthusiastically looking forward to hearing from you

protocentralashwin, 2018-04-06:
Merging this with issue #13

closed 2018-04-06 by protocentralashwin