Protocentral/protocentral_max30003 #10: No ECG signal

stanky95, 2017-12-13:
Hi,
I have bought a protocentral breakout board for MAX30003 recently and I have been struggling to get it up and running. A similar issue is occuring to me as the one that can be found on the link below:

https://github.com/Protocentral/protocentral_max30003/issues/6

I'm running an Arduino Leonardo with the latest Arduino IDE version. I have tried several versions from the your github repository with no substantial difference. I have wired everything according to your schematic, except for the MISO/MOSI/SCK pins, which are wired to the ICSP header since Leonardo has SPI pins available only there, as far as I know. I am also using the provided cable and electrodes. Everything is connected via usb to the laptop disconnected from the mains.

I have also detected that it is not an issue in the GUI, since I have tried bypassing it by printing to serial only the ECG data, and not the whole data header, and monitoring it with Serial Plotter - with the same result present.

(just to clarify, I am basically using this:
Serial.println(ecgdata);
instead of the for loop for printing to the serial)

Do you have any clue what might be happening?

VenkateshBhat, 2017-12-14:
@stanky95 
Try out this code and let me know. I have changed the FCLK pin to a dedicated PWM pin. 
Connect the FCLK to digital 9 of Lenardo rest all pins are the same.

[test.txt](https://github.com/Protocentral/protocentral_max30003/files/1557999/test.txt)

stanky95, 2017-12-14:
@VenkateshBhat 
I have tried the code you provided and swapped the pins:
When using this code, the output is completely constant, usually at zero. It does not matter if leads are connected or not.

I presume that it might have to do with the FCLK? Since the original code was driving the FCLK pin every 16us, which means the period is 32us, which gives a frequency of 31250 Hz, while it is specified in the MAX30003 datasheet that the frequency should be 32768 Hz. I'm not sure why it doesn't work with the PWM.

VenkateshBhat, 2017-12-14:
Something is wrong here, it doesn't give a flat line.  I checked with the Lenardo board and it is working fine for me with PWM.
Yes the clock should be 32.768KHz, since we can not generate exact clock using the timer prescalar we are using clock close to that. 
kindly share your connection setup and arduino sketch.

AliAlzuabi, 2023-09-02:
I have a question, what did you use guys to generate FCLK = 32768Hz for Maz30003 ?

protocentralashwin, 2023-09-02:
Hi @AliAlzuabi , in the latest versions of the boards, 32 KHz clock is generated by a onboard CMOS oscillator and that works fine. So you do not have to generate a separate clock for this.

AliAlzuabi, 2023-09-02:
Hello @protocentralashwin, are you suggesting that if I've used an ESP32 Dev module, and it has generated an FCLK of 32Khz, can this be worked as FCLK for the Max30003? No need for external crystal ?