Protocentral/protocentral_max30003 #11: Noisy data and unclear .csv readings.

rogzam, 2018-01-24:
Hello,

I've acquired your max30003 module and hooked up to an Arduino Uno and after several iterations I managed to get a decent Heart Rate signal. However:

Even though I've followed your instructions carefully the data is still really noisy,  I've tried several positions for the electrodes, staying still, unplugged everything else from the laptop (I found out that my HDMI monitor and laptop charger were inducing a lot of noise to the signal), etc. Is there any way to get a cleaner data? Something similar to what you present in your webpage?.. This is my cleanest reading so far:

[image]

Another thing that it's not fully clear to me is the wiring. Here you say that the module has a 3.3v pin, a 5v and a GND:

[image]

However, the product says GND - VCC - GND, so, which one is it? Should I plug both pins to the ground? should I plug one to 5v, one to 3.3v and the last one to GND? or just 5v and GND and leave the last one unplugged? 

[image]

Lastly, once I open the raw data I get these columns:

[image]

Can you please explain what do they mean? what are they measuring and what scale are there using? I understand that the first one is the timestamp, but, ECG?, SpO2?, respiration?

I'm attaching the .csv -> 
[log.csv.zip](https://github.com/Protocentral/protocentral_max30003/files/1660201/log.csv.zip)

Thanks for your help.

rogzam, 2018-01-31:
@VenkateshBhat, @protocentralashwin ... Any comments, ideas?

protocentralashwin, 2018-02-01:
@rogzam I'm sorry about the wrong documentation. Earlier we had 3v3 and VCC separately, but now we combined both together in the new version of the board, so the other pin is not connected to power, you can leave it unconnected or to ground.

As for the noise, It depends on a lot of things. Since this is only 2-electrode measurement, there is no third electrode for noise compensation. I would mainly check the power supply to make sure it is stable. What source are you using to power the board? Also, the electrodes cables and contact would be important. Please keep in mind that this IC was designed for wearable applications and you would get best results from a battery operated system.

As for the log file, I think the example code has some extra fields that we were using for testing, you can ignore those. We will correct the code.

rogzam, 2018-02-02:
@protocentralashwin Thanks for your answer.

I'm using a Lipo Battery to power the device and have tested different positions for the electrodes being the one recorded on the log I attached the cleanest one. Any recommendations in order to get a signal as clean as the one pictured on your webpage?

Also, can you give me a little more information on the data recorded on the csv?.  What's the unit of what is being shown on the file?

Thanks.

rogzam, 2018-02-07:
@VenkateshBhat, @protocentralashwin ?

iharshadec, 2018-02-23:
I am also seeing so much noise and signal is not as clear as shown on webpage. I am running board via Battery Only.

protocentralashwin, 2018-04-06:
@rogzam @iharshadec  Can you try powering the board separately through a battery? A bad arduino with which has a ground in common with the device can add noise. Since this is a 2-electrode ECG without a driven right leg (DRL), there is no way to drive back the common mode.

amir1994a, 2018-05-17:
@rogzam @iharshadec  were you able to fix your problem? I'm facing a noisy signal too...

iharshadec, 2018-05-17:
@amir1994a :
No, I am not able to get it even with separate battery.

amir1994a, 2018-05-17:
@iharshadec  
Are you using your own board? or the protocentral module?

rogzam, 2018-05-17:
@amir1994a @iharshadec I managed to make it work. I have the module working with a Nano and a Lipo. I found out that my monitor's power brick was causing some of the noise. Once I stepped away from the monitor I got a cleaner signal. However, is nothing like the one shown on Protocentral's video... Still disappointed of their product and they total lack of communication.