Protocentral/protocentral_max30003 #12: Max30003 problem with SPI

adelalfusail, 2018-02-27:
Hi, my project is patient Monitor I already bought max30003,afe4490,max3025 and I'm trying collect them in one arduino(mega 2025),everything is ok except when I'm connect SPI pins for max30003 and afe4490 the date is confuse  or maybe the is lost for max 30003 I have been trying to fix this problem almost two months and no response .
I just need to what is the problem.
 notice :the main problem is when I connect miso for both device

georod99, 2018-04-05:
Hi, I'm a retired physics teacher, thou I worked in the physics labs at Havana University for many years.
Could you make the Max30003 work? I haven't. Can you post pictures of what you did?
I'm trying to get HRV real-time. What is your objective?
More after you answer. Thanks.

protocentralashwin, 2018-04-06:
@adelalfusail Please make sure that the CS lines are separate and only one device is selected at a time when the SPI bus is used.