Protocentral/protocentral_max30003 #16: Bad signal and unit question?

PaginaRice, 2018-05-02:
Hello protocentral community,
I have been working on a project using the MAX30003 for several weeks now. Our signal is very unclear, (see attached image). Additionally, I am struggling to use the .csv file to troubleshoot because I don't know what the units are for the ecg data, and  the "timestamp" column is showing the same time for every single datapoint.

Does anyone have any insight on how to improve our signal and what the output data even means (units-wise)? Lastly, I noticed on the datasheet that it said filtering was an option, does that mean we have to opt to turn it on or off somewhere within the code? 

[image]

[image]

Any help would be greatly appreciated!
Thanks!

adelalfusail, 2018-05-10:
there is some reasons you need to check:
 1-you need to make sure to pin connection 
2-you need to discharge computer from charger 
3-stay far from power source 60/120 HZ

amir1994a, 2018-05-17:
@PaginaRice Hi, I'm facing the same problem for a long time! Have you been able to fix it?

adelalfusail, 2018-07-14:
Can you explain more to understand your problem????

joicetm, 2018-10-05:
are you able to fix it?
if not, you try with the latest [commit](https://github.com/Protocentral/protocentral_max30003)

joicetm, 2018-11-24:
Hi,
I am closing this issue for now, please feel free to open if its not fixed.

closed 2018-11-24 by joicetm