Protocentral/protocentral_max30003 #18: reading speed

agentff6600, 2020-09-24:
Hi!

I might have a question that seems very obvious for you. But I guess I need to ask somebody.

I have a hard time to figure out what the update speed is. When I plug in the Arduino and look at the serial monitor (or your Gui) it seems like that the data is sent in 1 sec burst.

Is that correct? How could I change that speed?

Maximum20, 2021-05-27:
Did you get it? How to change the speed?

joicetm, 2021-05-27:
Hi,

You meant the speed at which the ECG signal is streamed ?
It samples the data in every 8ms (It is configured with a sampling rate of 125 Hz), and sends it in real time. as you can see in the arduino plotter example, it sends the data serially in every 8ms.

Why do you want to increase the speed ?

Maximum20, 2021-05-27:
Sorry, I don't need to change the speed actually, I need to increase the sample rate because the signal is not detailed enough. Can I do it?

joicetm, 2021-05-28:
Yes,

It has 3 different sampling rates possible, (128, 256, 512). you can do this by writing to CNFG_ECG Register
[image]

There is a function in the library to set the sampling rate, call this after max30003_begin() 
```
void MAX30003::max30003SetsamplingRate(uint16_t samplingRate)
```

It is not tested but you may try it out.
you will also have to adjust the delay in the loop according to the configured sampling rate.

joicetm, 2021-05-29:
fyi,

You can increase the gain also If you are looking to increase the signal quality.
The default gain in the example configuration is only 20.

Maximum20, 2021-05-29:
@joicetm I've tried MAX30003_Reg_Write(CNFG_ECG, 0x805011); 
If I get right 0x8050**11** (11) mean to set ECG channel Gain to 160V/V isn't it?
I get more information from the signal but I  notice that some heart beats still don't appear on the graph . 

if i use MAX30003_Reg_Write(CNFG_ECG, 0x005011);  I get more noisy signal
And what does the gain mean? How does it work?

joicetm, 2021-05-31:
No,
Its in binary, you will have to write the corresponding hex/decimal value.

Please try 0x835000 for 128 sampling rate and 160 gain. 
Keep the hpf and lpf as it will help to reduce the noise.