Protocentral/protocentral_max30003 #20: Arduino UNO and Max30003 problem

Maximum20, 2021-01-09:
**Serial Port of arduino always shows me 0**. Initialization is successful as I get it right but I get only zeros in the serial port. I've tried all the three example codes and no one shows me the values opposite to 0. 
 **And also photocentralview is always shutting down and creates a .txt file when I press Start**. (I correctly chose the port and board name).
 What's the problem can it be?

protocentralashwin, 2021-01-09:
@Maximum20 

Can you please let us know which GUI you're trying to use and which Arduino sketch is loaded on the Arduino Uno?

Also, your operating system would also help.

Maximum20, 2021-01-09:
> Can you please let us know which GUI you're trying to use and which Arduino sketch is loaded on the Arduino Uno?
> Also, your operating system would also help.

Thank you for reply. I use pc-openview-windows64 on Windows 10. I left here a log file which this GUI creates after shutting down.
I tried to use three example codes. it's ECG-stream-openview, ECG-stream-Arduino-Plotter and Computation-only. 
I use Arduino 1.8.13. 

[hs_err_pid14980.log](https://github.com/Protocentral/protocentral_max30003/files/5791414/hs_err_pid14980.log)

I made a video when I showed my issue. The link is below
https://youtu.be/aqnpOfYB1pk

protocentralashwin, 2021-01-09:
Thank you sending all the details, it looks like some kind of Java access violation with the OpenView software. We will check on this and get back to you.

Meanwhile, the other two examples, especially Arduino plotter, should print the waveform in the Arduino Serial plotter utility. Can you make sure that the baud rate there is set at 57600 and check?

Maximum20, 2021-01-09:
> Meanwhile, the other two examples, especially Arduino plotter, should print the waveform in the Arduino Serial plotter utility. Can you make sure that the baud rate there is set at 57600 and check?

Yes, of course. Here is another video, when I use it with Arduino Serial plotter. But I get 0 all the time :(
https://youtu.be/5lhlD8ywdR4

protocentralashwin, 2021-01-09:
@MohammedHM will help you out with this, first thing on Monday.

To me, at first, it looks like some connection issue (between the breakout and the Arduino). Sometimes, the wires can cause signal loss, especially for SPI data lines. 

The OpenView problem and this are not related, seems like two different issues.

Maximum20, 2021-01-09:
> @MohammedHM will help you out with this, first thing on Monday.
>
> To me, at first, it looks like some connection issue (between the breakout and the Arduino). Sometimes, the wires can cause signal [...]

Thank you for your reply. I'll wait for  @MohammedHM help so much

MohammedHM, 2021-01-11:
Hi @Maximum20, could you please share the image of your setup, interface between Arduino UNO and max30003 breakout.

Maximum20, 2021-01-11:
> Hi @Maximum20, could you please share the image of your setup, interface between Arduino UNO and max30003 breakout.

Hi. Thank you for your reply Here is it
https://imgur.com/a/kmvyv3w

SCK(SCLK) - D13 
MISO - D12
MOSI - D11
CSB(CS) - D7
FCLC - D6
VCC - 3.3V
GND - GND

Maximum20, 2021-01-11:
[image]

joicetm, 2021-01-12:
@Maximum20 Would please remove the FCLK connection and try once. also please connect INT1 pin to D2(This is required for computation only example)

  |MAX30003 pin label| Pin Function         |Arduino Connection|
  |----------------- |:--------------------:|-----------------:|
  | MISO             | Slave Out            |  D12             |
  | MOSI             | Slave In             |  D11             |
  | SCLK             | Serial Clock         |  D13             |
  | CS               | Chip Select          |  D7              |
  | VCC              | Digital VDD          |  +5V             |
  | GND              | Digital Gnd          |  Gnd             |
  | FCLK             | 32K CLOCK            |  -               |
  | INT1             | Interrupt1           |  02              |
  | INT2             | Interrupt2           |  -               |

Maximum20, 2021-01-12:
> @Maximum20 Would please remove the FCLK connection and try once. also please connect INT1 pin to D2(This is required for computation only example)
>
> MAX30003 pin label	Pin Function	Arduino Connection [...]

@joicetm , Need I use 3.3V VCC or 5V? Doesn't it matter?

joicetm, 2021-01-12:
you should use 5V.

Maximum20, 2021-01-12:
@joicetm 
I've checked if the library is 1.2.0 and it is 1.2.0. 
when I'm trying to use "Example3-computation-only" I get this:

max30003 is ready
Rev ID:  80
Max30003 ID Success
Initialising the chip ...
And nothing else. When I use "Example2 stream Arduino plotter" I get the same but after "Initialising the chip ..." I get zeros in the serial port of arduino.

Here are the photos of my breakout board:
[image]
[image]

joicetm, 2021-01-12:
@Maximum20 where did you buy this from ? 
This is not a protocentral product, i am not sure if the board has any design/hardware flaws. please contact the manufacturer for more details.

you can buy one genuine protocentral max30003 breakout  from our [website](https://protocentral.com/product/protocentral-max30003-single-lead-ecg-breakout-board-v2/).

Maximum20, 2021-01-12:
> @Maximum20 where did you buy this from ?
> This is not a protocentral product, i am not sure if the board has any design/hardware flaws. please contact the manufacturer for more details.

Oh, okay. I got it. Thank you

closed 2021-01-12 by joicetm