Protocentral/protocentral_max30003 #23: Sporadic 20BPM then 240BPM

elliotglynn1, 2021-08-10:
Is there a specific set of registers that allow for the cleanest data struggling to find the issue with the rtor readings coming from the MAX30003?
At the moment everything is set to default, with calibration and leads off detection both switched off.
Any help would be greatly appreciated!

joicetm, 2021-08-11:
Hi,

I didn't fully understand your question, can you please tell me what is your exact issue?
Are you facing issues with rtor reading ? which code did you try,  Example3-Computation-only ? are you not getting the reading?

Our example arduino code is simple generic one written for demonstration purposes and on each example we have only tried to demonstrate that particular feature.

You will have to configure EN_INT and CNFG_GEN registers If you want to enable lead detection.

What do you want to do with the calibration signal? you make small chnages to the ECG-stream example make it stream the calibration signal.

elliotglynn1, 2021-08-11:
Sorry realised after I posted this it was for an arduino board and I’m not using one. I was just wondering if there was a generic set of registers to get the cleanest signal and if there was something I may have missed.

I am using the max30003 connected to a chest strap with 2 leads and the firmware is working perfectly with the calibration signal but when switched off the actual rtor readings are sporadically noisy. Also tried to change the RTOR algorithm settings but had no luck.

Thanks for all your help!