Protocentral/protocentral_max30003 #24: Connection issue

matieu, 2022-09-22:
Hi,
I am trying to get a signal from the MAX30003. I followed the code example but I am stuck here: 

Serial.println("Failed to read ID, please make sure all the pins are connected");

I spent a few hours trying to debug this with not luck.

Please see my connections in the image. (I'm using the code sample from this repo)
-Am I doing something wrong?
-Is the card working? How can I test it?

I really want to make it work. Could please help me?

Many thanks!
[image]

MohammedHM, 2022-09-23:
Hi Matieu,

It appears that there is a contact issue between the header and the MAX30003 breakout board. Your interface between Arduino UNO and MAX30003 via breadboard is correct.

Can you solder the headers to the pins on the MAX30003 breakout board and try uploading the example code?

matieu, 2022-09-23:
Indeed, it resolved the issue.

Thanks you so much!

closed 2022-09-23 by matieu